schnatterer/rooted-graphene #1: Also build unrooted image?

schnatterer, 2024-03-28:
This would allow for enabling/disabling root via OTA.

Use subfolders in Github page, so switching is possible by changing URL in custota?

pascallj, 2024-04-07:
Please be aware that Custota doesn't work on an unrooted OTA by default. Unless you somehow modify the system image to include the Custota stuff into the rom, but it doesn't seem like that's what you are trying to do from an initial glance at the repo.

So once you go unrooted, you can't update anymore with Custota. You can by flashing with ADB however.

schnatterer, 2024-04-07:
Thanks @pascallj, I aware of this. I just like the idea of being able to simply unroot. Even if it's just to try out if you're banking app is crashing due to root detection or some other reason :roll_eyes: 
Yes, for rooting again later and for OTA, you'll have to `adb sideload` first.

closed 2024-04-18 by schnatterer

xabolcs, 2024-04-19:
> ... Custota doesn't work on an unrooted image by default. Unless you somehow modify the system image to include the Custota stuff into the rom ...

For the record: an `ext4` formatted `system.img` could be easily modified with `mount`, `resize2fs` and `e2fsck` tools. More details in the [**avbroot** discussions](https://github.com/chenxiaolong/avbroot/discussions/260#discussioncomment-8792626).

I thought about adding Custota to (a rootless) system already. :slightly_smiling_face: 
The missing part was this `rooted-graphene` functionality!

So thanks @schnatterer for sharing! :pray:

schnatterer, 2024-04-19:
@xabolcs you're welcome 😊! 

So with these simple commands plus adding the custota apk to the system apps, custota could be added to the rootless OTA without breaking AVB?

Thanks for pointing this out!
Did you try if this theory holds in practice?
Do you happen to have a snippet at hand? 😁

xabolcs, 2024-04-19:
> ... custota could be added to the rootless OTA without breaking AVB?
> Did you try if this theory holds in practice?

I never tried adding Custota. But had success with adding GApps! :muscle:
Just read that discussion from the beginning!

>  Do you happen to have a snippet at hand? 😁

Nay! I was too lazy to cook a script from my command history.
But that discussion have some! :ok_hand:

xabolcs, 2024-04-20:
While adding apps to `system.img` isn't a big deal, but all the `*.sh` in `Custota-release.zip` make me worry. :confused:

schnatterer, 2024-04-21:
In theory we could just run those on the mounted system img. In practice, they surely have some dependencies to magisk. Probably should all Andrew for the best way to include custota into an ota.