schnatterer/rooted-graphene #8: Some questions about the project

silverwings15, 2024-07-25:
Hiya, I just managed to successfully setup GOS + avbroot with Kitsune Magisk last week :D just discovered your project and i have some questions as follows: 

- How are you able to use vanilla Magisk? Afaik only HuskyDG's Kitsune and Pixincreate's Magisk are compatible with GOS, and I don't see this mentioned in the instruction
- I see that only Shiba is supported currently, will OTA for other devices be available in the future?
- Is the process automated? If not, for how long do you think this project can/will be maintained? 

Thanks for your time!

schnatterer, 2024-07-25:
Hey @silverwings15 ,

I'll do my best to answer

> How are you able to use vanilla Magisk? Afaik only HuskyDG's Kitsune and Pixincreate's Magisk How are you able to use vanilla Magisk? Afaik only HuskyDG's Kitsune and Pixincreate's Magisk are compatible with GOS, and I don't see this mentioned in the instructionare compatible with GOS, and I don't see this mentioned in the instruction

I think the Readme has an answer to that:
> OS and root work in general. However, zygisk does not (and likely never will) work, leading to magisk being easily discovered by other apps and lots of banking apps not working.

Do you want to know anything else in particular?
You could create a fork that uses one of the alternatives you mentioned. I won't because relying on root and a 3rd party software for repacking GOS (avbroot) is bad enough for my software supply chain. I don't want to give the keys to my device to some arbitrary software. Too bad magisk refuses to accept patches for GOS support...

> I see that only Shiba is supported currently (and Oriole up until 3 months ago), will OTA for other devices be available in the future?

First we would have extend the code to allow for multiple devices. 
Either the script (more efficient), or the github action (probably less effort, I tried it but it failed: 4c109d8) 

Then it would be rather simple, you only need a device and a magisk preinit string that you can find out on the device using the magisk app (see Readme).

I'm a bit hesitant to build multiple devices though, because I'm not sure how much space I'm allowed to consume for my github releases. At the moment for each release two files (magisk and rootless) with about 1.1 GB are created per file.

You could fork the repo and just change the github action to build your device with your magisk string. This way you would also own your own signing key (which you would of course have to create first).

> Is the process automated? If not, for how long do you think this project can/will be maintained?

The process of releasing as well as updating depencenies like magisk, avbroot and custota is automated.
I use this myself at the moment and hope to be using my pixel for the time it is supported, i.e. five more years.
However, this project very much relies on avbroot. So, if avbroot should be no longer supported, this project won't either, probably.
I think especially major android releases and potentially every magisk version will require changes there

Also the main reasons for me to need root is backup. GOS might provide a better solution than seedvault one day (hopefully), which might tip me towards stop requiring root in favor of more security.

So, I hope the fun will last for a while, but no promises there. 

HTH

silverwings15, 2024-07-26:
appreciate the answers!

closed 2024-07-26 by silverwings15