schnatterer/rooted-graphene #30: Kernelsu

radioaddition, 2024-10-15:
I realize you mentioned kernelsu being much more difficult to patch in than magisk, but there seems to already be [a repository](https://github.com/cawilliamson/rooted-graphene) that does most of the patching, it just doesn't have builds through github actions, and doesn't support all gos devices as is. Either way, might be worth a look

schnatterer, 2024-10-16:
Thank you for pointing this out. For now I gave up on kernelsu, also because I wasn't sure about kernelsu being completely trustworthy, as pointed out in the Readme.
I might look into this later, but at first glance the scripts in the repo you mentioned seem to build kernel and GOS from source instead of patching them,. Building from source requires a lot of ram and cpu, which cannot be done with the rather scarce resources of free github actions.

radioaddition, 2024-10-16:
I don't mean to reopen the issue, but what are your concerns with kernelsu? I've been using it and I wanna make sure I shouldn't be worried about something. I don't mean this to start an argument or anything, I'm just genuinely curious as to what I should look into before I continue to use it

schnatterer, 2024-10-17:
You're welcome to reopen it. Maybe some else is interested in working on it.

My concerns follow [a statement by the developer of avbroot](https://github.com/chenxiaolong/avbroot/discussions/267) , about parts of kernelsu not being open source. On top of that I just don't have a lot of spare time and have to choose wisely. Spending time one something complex with might not be trustworthy doesn't put it very high up on my list of priorities.

radioaddition, 2024-10-17:
It seems like that statement is talking specifically about the kernel module implementation, which would mean that is building the kernel through actions would work, that method would be fully open source

schnatterer, 2024-10-17:
As stated, I haven't had the time to dig deep enough into the world of kernelsu to be able to judge what is right here. 
Maybe @chenxiaolong is able to join this discussion at this point?

chenxiaolong, 2024-10-17:
> seem to build kernel and GOS from source instead of patching them

I'm pretty sure this is necessary to use KernelSU with GrapheneOS. They make a lot of improvements to the kernel that userspace depends on, so the system can't boot with a GKI kernel (like the prebuilt ones provided by KernelSU).

(The closed-source kernel module method won't work either because the GrapheneOS kernel enforces signature verification for kernel modules. To add a new key to the list of trusted keys, you would have to rebuild the kernel anyway.)

DADA30000, 2025-11-04:
so hey, sorry for disturbing this old issue, but does this linked method
https://github.com/cawilliamson/rooted-graphene
still works? I'm personally fine with building grapheneos from scratch, and don't want to use magisk as it's detected by many apps, and most of actually working hiding helpers are closed source (shamiko for example), so I want to use KSU Next + SuSFS.
Just want to make sure it's even working, to not waste dozens of hours building something that people know doesn't work.