schnatterer/rooted-graphene #31: any plans on supporting pixel 9?

FinGu, 2024-10-16:
Or maybe a guide to do so? I have a Pixel 9 Pro, given build instructions i could probably compile it and test locally..

schnatterer, 2024-10-16:
In general, I'm not planning to support many devices myself, as each release uses a lot of gigabytes and I'm not sure at what point github will charge me for it. That said, I might acquire a Pixel 9 Pro, but not no guarantees here. 
Also, I'm not sure if this will keep working after major android version changes. As AOSP 15 has just been released, we'll know soon.

I wouldn't build locally. 
You could fork the repo and just change the github action to build your device with your magisk string. This way you would also own your own signing key (which you would of course have to create first).

What you need to do is find out the magisk preinit string, see [Readme](https://github.com/schnatterer/rooted-graphene/tree/8c4ff0926183b7ef483d4a76867de5f0078841ca?tab=readme-ov-file#magisk-preinit-strings).. 
Then add your device to the github action: https://github.com/schnatterer/rooted-graphene/blob/8c4ff0926183b7ef483d4a76867de5f0078841ca/.github/workflows/release-multiple.yaml#L21

schnatterer, 2024-10-16:
If you know anything about limits in github releases let me know. 
Also I'd be happy to add the pixel 9 and its preinit string to the code in comments. Who knows, maybe I can support multiple devices at some point.

FinGu, 2024-10-16:
Seems like this is it 
[image]
Built it on my repo, will be trying it later

FinGu, 2024-10-16:
btw, shouldn't this be $partition_name ?
[image]

schnatterer, 2024-10-17:
> Seems like this is it ![grafik](https://private-user-images.githubusercontent.com/50637180/377198936-39ce3d49-6026-41cc-9c13-fa2f12069b4e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjkxNDUwOTEsIm5iZiI6MTcyOTE0NDc5MSwicGF0aCI6Ii81MDYzNzE4MC8zNzcxOTg5MzYtMzljZTNkNDktNjAyNi00MWNjLTljMTMtZmEyZjEyMDY5YjRlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDE3VDA1NTk1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY3ZTBlMDVmY2I5OTRkNDE5ZjViNDc0MTA2ZWQzNTcyNzM0NTdmZWMzYzMzYWVlMTY3OTIwNDQ1NDRmOGNiYjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.OYTF2k-4bwL466Xtt75gnZ4WY4HUgpDpA3NWVhysUjk) Built it on my repo, will be trying it later

My concern is about storage in github releases, where the OTAs are uploaded to

schnatterer, 2024-10-17:
> btw, shouldn't this be $partition_name ? ![grafik](https://private-user-images.githubusercontent.com/50637180/377201622-162d4556-dde2-4fd3-b605-9e5c7569d67a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjkxNDUwOTEsIm5iZiI6MTcyOTE0NDc5MSwicGF0aCI6Ii81MDYzNzE4MC8zNzcyMDE2MjItMTYyZDQ1NTYtZGRlMi00ZmQzLWI2MDUtOWU1Yzc1NjlkNjdhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDE3VDA1NTk1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThiNzE4N2FlNjg2NTcyMjc3MmUzMWRmMzA4MDM2NjA0ZTVlNDM3NGMyODA5NTYwNjdlYzcyMTc5NTkyNDIxNTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.VzpQBLgK4AWX29JZY0fwwfTEz_J79T8Dhppn--hapR4)

Good catch! 

By now, the instructions could even be simpler.
Have a look at https://github.com/chenxiaolong/avbroot/tree/v3.8.0?tab=readme-ov-file#initial-setup
Steps 4 to 6.
Would you create a PR for this?
I haven't set up a new device since the instructions were updated.

FinGu, 2024-10-17:
> Good catch!
>
> By now, the instructions could even be simpler. Have a look at https://github.com/chenxiaolong/avbroot/tree/v3.8.0?tab=readme-ov-file#initial-setup Steps 4 to 6. Would you create a PR [...]

Sure, i'll make one for it later

FinGu, 2024-10-17:
> My concern is about storage in github releases, where the OTAs are uploaded to

Seems like it's okay https://docs.github.com/en/repositories/releasing-projects-on-github/about-releases
Check storage and bandwidth quotas

FinGu, 2024-10-18:
Should be done, it just needs release multiple to be triggered again i guess

closed 2024-10-18 by FinGu

FinGu, 2024-10-18:
I am, it's just because of zygisk support mainly rn. I probably don't need it tho