schnatterer/rooted-graphene #56: Issue with fastboot flashball last version

accltza, 2025-01-08:
Hello,
thanks a lot for your work, it seem there is some issue during flastboot step.

Checking 'product'                                 OKAY [  0.000s]
Setting current slot to 'a'                        OKAY [  0.072s]
Sending 'boot_a' (65536 KB)                        OKAY [  2.296s]
Writing 'boot_a'                                   OKAY [  0.075s]
Sending 'vbmeta_a' (8 KB)                          OKAY [  0.001s]
Writing 'vbmeta_a'                                 OKAY [  0.002s]
Resizing 'product_a'                               FAILED (remote: 'Invalid command resize-logical-partition:product_a:0')
fastboot: error: Command failed

schnatterer, 2025-01-08:
Which device and version?
It works for me with 2024123000 for caiman.
Can you find something about this issue in avbroot's issues!

accltza, 2025-01-08:
Pixel 6A , yes with the  bluejay-install-2024123000  version OTA realease corresponding, first part done with web install.
You surely right will dig that way.

just-for-death, 2025-01-20:
yes it works for me @schnatterer on my device   I couldn't daily drive it as I needed zygisk support but its an awesome project

schnatterer, 2025-01-21:
Thanks for confirmation @just-for-death .
I will close this for now.
 @accltza you're free to reopen or discuss when there is any news.

closed 2025-01-21 by schnatterer