schnatterer/rooted-graphene #76: failed to load key

OwOday, 2025-02-27:
Hi, I am running the generate key and create and release commands as shown in the scripts section of the readme. 

command:
GITHUB_TOKEN=no SKIP_CLEANUP=true GITHUB_REPO=OwOday/rooted-graphene DEVICE_ID=felix MAGISK_PREINIT_DEVICE=sda10 bash -c '. ./rooted-ota.sh && createRootedOta'

output:
```
[INFO] Packing AVB image: /tmp/tmpklt60bw1/images/vendor_boot.img
  0.030s  INFO Leaving AVB header unsigned
[INFO] Patching OTA with replaced images: system, vendor, vendor_boot: .tmp/felix-2025021100-ca0b105-magisk-v28.1-dirty.zip
  0.000s ERROR Failed to load key: "avb.key"

Caused by:
    0: Failed to read file: "avb.key"
    1: No such file or directory (os error 2)
[ERROR] Failed to patch OTA
Traceback (most recent call last):
  File "/app/.tmp/my-avbroot-setup/patch.py", line 340, in main
    run(args, Path(temp_dir))
    ~~~^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.tmp/my-avbroot-setup/patch.py", line 308, in run
    external.patch_ota(
    ~~~~~~~~~~~~~~~~~~^
        args.input,
        ^^^^^^^^^^^
    ...<5 lines>...
        args.patch_arg,
        ^^^^^^^^^^^^^^^
    )
    ^
  File "/app/.tmp/my-avbroot-setup/lib/external.py", line 97, in patch_ota
    subprocess.check_call(cmd)
    ~~~~~~~~~~~~~~~~~~~~~^^^^^
  File "/usr/local/lib/python3.13/subprocess.py", line 421, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['avbroot', 'ota', 'patch', '--input', PosixPath('.tmp/felix-ota_update-2025021100.zip'), '--output', PosixPath('.tmp/felix-2025021100-ca0b105-magisk-v28.1-dirty.zip'), '--key-avb', PosixPath('avb.key'), '--key-ota', PosixPath('ota.key'), '--cert-ota', PosixPath('ota.crt'), '--magisk', '.tmp/magisk-v28.1.apk', '--magisk-preinit-device', 'sda10', '--replace', 'vendor_boot', PosixPath('/tmp/tmpklt60bw1/images/vendor_boot.img'), '--replace', 'system', PosixPath('/tmp/tmpklt60bw1/images/system.img'), '--replace', 'vendor', PosixPath('/tmp/tmpklt60bw1/images/vendor.img')]' returned non-zero exit status 1.
```

schnatterer, 2025-02-28:
You can either work around this bug using `KEY_AVB`, `KEY_OTA` and `CERT_OTA` or try #77 and tell me if this fixes it 🙂

schnatterer, 2025-03-13:
I'll interpret the thumbs up as fixed.

closed 2025-03-13 by schnatterer