schnatterer/rooted-graphene #82: Bootloop on caiman (Pixel 9 Pro)

diwangs, 2025-03-13:
I've been using `rooted-graphene` for a while and it worked great for my scenario. However, since the 20250211 release or so, I can't update to the new rooted version without encountering a bootloop. Switching to the rootless version seems to alleviate the issue, but I'd very much like to get root back if I could. I tried to look at the CustOTA log but they seem to only take logs on the OTA update process.

Is there any known bootloop issue and/or fix for Pixel 9 Pro?

Thank you in advance!

schnatterer, 2025-03-13:
I have my suspicions 🤔
Around this time I started integrating custota into the OTA.
https://github.com/schnatterer/rooted-graphene?tab=readme-ov-file#2025021100

If you had it installed as magisk module first there might be trouble.
I had problems on one device. I don't  exactly recall exactly what it was and how I fixed it. I don't think it was a bootloop.  Can you go back to the other partition or last version with root and uninstall the custota magisk module?

diwangs, 2025-03-15:
Unfortunately I have since updated my phone to the newer rootless version 😅 

If the culprit is truly the custota magisk module, then I'm afraid the only way to modify them is through re-flashing the magisk version, since you need root permission to delete magisk module (even through ADB). But then again, you don't even encounter the problem so perhaps the problem is something else entirely?

Before I try to re-flash my phone, do you have other related suspicions that I could look into?

diwangs, 2025-03-16:
Good news, I think you might be right all along that the culprit is the custota module. I can't really pinpoint what fixed it exactly, but this is what I did:

1. Update the ROM to `250314` magisk version
2. Boot into safe mode (disabling installed app)
3. Install the Magisk app
4. Enable USB debugging
5. Connect the phone to a PC with ADB and reboot
6. While the phone is rebooting, type `adb wait-for-device shell magisk --remove-module` ([source](https://didgeridoohan.com/magisk/Magisk#hn_Module_causing_issues_Magisk_functionality_bootloop_loss_of_root_etc))
7. The phone will automatically reboot once again, but after that the ROM works as expected

As expected, all the Magisk modules will be gone, so you'll need to reinstall modules as needed.

Thank you for the pointer and for all the good work!

closed 2025-03-16 by diwangs

schnatterer, 2025-03-16:
I am glad to hear that you escaped the boot loop!
Great work and thanks for sharing. 
TIL about calling magisk on boot and `adb wait-for-device` 💡