schnatterer/rooted-graphene #93: Device corupted too

orphen92, 2025-04-02:
Hi i have follow all the step but my google pixel 6 was corrupted each time.

This is all my step and logs:

Install **GraphenOs** **oriole-install-2025032500.zip** with webinstaller

Download **oriole-2025032500-1c3df7a-magisk-v28.1**

download **avbroot-3.14.0-x86_64-pc-windows-msvc**

Use windows cmd prompt line as admin in avbroot folder

**Extract partition images:**
.\avbroot.exe ota extract --input "C:\Users\sylva\Downloads\avbroot-3.14.0-x86_64-pc-windows-msvc\oriole-2025032500-1c3df7a-magisk-v28.1.zip" --directory extracted --fastboot

**set variable:**
set ANDROID_PRODUCT_OUT=C:\Users\sylva\Downloads\avbroot-3.14.0-x86_64-pc-windows-msvc\extracted

**Flash the partitions:**
fastboot flashall --skip-reboot

--------------------------------------------
Bootloader Version...: slider-15.2-12893632
Baseband Version.....: g5123b-145971-250103-B-12866815
Serial Number........: 25271FDF6001MS
--------------------------------------------
Checking 'product'                                 OKAY [  0.000s]
Setting current slot to 'a'                        OKAY [  0.093s]
Sending 'boot_a' (65536 KB)                        OKAY [  1.603s]
Writing 'boot_a'                                   OKAY [  0.278s]
Sending 'vbmeta_a' (8 KB)                          OKAY [  0.001s]
Writing 'vbmeta_a'                                 OKAY [  0.006s]
Sending 'vendor_boot_a' (65536 KB)                 OKAY [  1.588s]
Writing 'vendor_boot_a'                            OKAY [  0.265s]
Rebooting into fastboot                            OKAY [  0.000s]
< waiting for any device >
Resizing 'system_a'                                OKAY [  0.007s]
Sending sparse 'system_a' 1/6 (262108 KB)          OKAY [  6.601s]
Writing 'system_a'                                 OKAY [  1.259s]
Sending sparse 'system_a' 2/6 (262128 KB)          OKAY [  6.516s]
Writing 'system_a'                                 OKAY [  1.178s]
Sending sparse 'system_a' 3/6 (262112 KB)          OKAY [  6.496s]
Writing 'system_a'                                 OKAY [  1.277s]
Sending sparse 'system_a' 4/6 (262108 KB)          OKAY [  6.505s]
Writing 'system_a'                                 OKAY [  1.226s]
Sending sparse 'system_a' 5/6 (262128 KB)          OKAY [  6.564s]
Writing 'system_a'                                 OKAY [  1.235s]
Sending sparse 'system_a' 6/6 (162000 KB)          OKAY [  4.054s]
Writing 'system_a'                                 OKAY [  0.773s]
Finished. Total time: 67.605s

fastboot reboot-bootloader

fastboot erase avb_custom_key

Erasing 'avb_custom_key'                           (bootloader) avb custom key: erase done
OKAY [  0.024s]
Finished. Total time: 0.026s

curl -s https://raw.githubusercontent.com/rooted-graphene/ota/refs/heads/main/avb_pkmd.bin > avb_pkmd.bin

fastboot flash avb_custom_key avb_pkmd.bin

Warning: skip copying avb_custom_key image avb footer (avb_custom_key partition size: 0, avb_custom_key image size: 1032).
Sending 'avb_custom_key' (1 KB)                    OKAY [  0.000s]
Writing 'avb_custom_key'                           (bootloader) avb custom key: flash done
OKAY [  0.059s]
Finished. Total time: 0.064s

fastboot reboot

Your device is corrupt

Can you help me please?

FinGu, 2025-04-02:
Could you try flashing this ota instead? https://github.com/rooted-graphene/ota/releases/download/2025032500/oriole-2025032500-7c061ff-magisk-v28.1.zip

orphen92, 2025-04-03:
> Could you try flashing this ota instead? https://github.com/rooted-graphene/ota/releases/download/2025032500/oriole-2025032500-7c061ff-magisk-v28.1.zip

Thats work great with this 

thanks

schnatterer, 2025-04-04:
@orphen92 please note this: https://github.com/schnatterer/rooted-graphene/issues/89#issuecomment-2773500629

Lets continue possible discussions in #89

closed 2025-04-04 by schnatterer