schnatterer/rooted-graphene #96: caiman 2025040400: doesn't boot

mocarela, 2025-04-06:
```
e:\Google Pixel 9 Pro>fastboot flashall --skip-reboot
--------------------------------------------
Bootloader Version...: ripcurrentpro-15.2-12817298
Baseband Version.....: g5400c-241205-250127-B-12971716
Serial Number........: 46211FDAP004VH
--------------------------------------------
Checking 'product'                                 OKAY [  0.001s]
Setting current slot to 'a'                        OKAY [  0.131s]
Sending 'boot_a' (65536 KB)                        OKAY [  1.528s]
Writing 'boot_a'                                   OKAY [  0.120s]
Sending 'init_boot_a' (8192 KB)                    OKAY [  0.188s]
Writing 'init_boot_a'                              OKAY [  0.017s]
Sending 'vbmeta_a' (8 KB)                          OKAY [  0.001s]
Writing 'vbmeta_a'                                 OKAY [  0.008s]
Sending 'vendor_boot_a' (65536 KB)                 OKAY [  1.535s]
Writing 'vendor_boot_a'                            OKAY [  0.117s]
Rebooting into fastboot                            OKAY [  0.001s]
< waiting for any device >
Resizing 'system_a'                                OKAY [  0.011s]
Sending sparse 'system_a' 1/6 (262112 KB)          OKAY [  6.405s]
Writing 'system_a'                                 OKAY [  0.523s]
Sending sparse 'system_a' 2/6 (262128 KB)          OKAY [  6.352s]
Writing 'system_a'                                 OKAY [  0.595s]
Sending sparse 'system_a' 3/6 (262112 KB)          OKAY [  6.382s]
Writing 'system_a'                                 OKAY [  0.634s]
Sending sparse 'system_a' 4/6 (262120 KB)          OKAY [  6.383s]
Writing 'system_a'                                 OKAY [  0.611s]
Sending sparse 'system_a' 5/6 (255008 KB)          OKAY [  6.185s]
Writing 'system_a'                                 OKAY [  0.581s]
Sending sparse 'system_a' 6/6 (20732 KB)           OKAY [  0.524s]
Writing 'system_a'                                 OKAY [  0.106s]
Finished. Total time: 74.366s

e:\Google Pixel 9 Pro>fastboot reboot-bootloader
Rebooting into bootloader                          OKAY [  0.001s]
Finished. Total time: 0.004s

e:\Google Pixel 9 Pro>fastboot erase avb_custom_key
Erasing 'avb_custom_key'                           (bootloader) avb custom key: erase done
OKAY [  0.029s]
Finished. Total time: 0.036s

e:\Google Pixel 9 Pro>fastboot flash avb_custom_key avb_pkmd.bin
Warning: skip copying avb_custom_key image avb footer (avb_custom_key partition size: 0, avb_custom_key image size: 1032).
Sending 'avb_custom_key' (1 KB)                    OKAY [  0.002s]
Writing 'avb_custom_key'                           (bootloader) avb custom key: flash done
OKAY [  0.043s]
Finished. Total time: 0.063s

e:\Google Pixel 9 Pro>fastboot reboot
Rebooting                                          OKAY [  0.000s]
Finished. Total time: 0.002s
```
In addition, bootloader shows a red warning to press a key in order to continue.

schnatterer, 2025-04-06:
No error message?

mocarela, 2025-04-06:
No... no error during the procedure. If repeating the procedure, sometimes the same bootloader warning is shown during reboot to fastboot mode.

persongyu, 2025-04-06:
chiming in to say this happened to me too. ive reflashed with slight changes about 400 times. pressing the power button doesnt bypass anything it just sends me to a google logo so my phone can become a space heater

schnatterer, 2025-04-07:
@persongyu @mocarela
Did you both do an initial install or OTA update? 
Does this also happen with the rootless variant?
If so you could update from there to rooted via OTA.

If not, I'll have to think about removing the modules,which would decrease the "(un) rooting via OTA" UX 😐

mocarela, 2025-04-07:
I managed to handle it by first flashing the rootless variant with fastboot (as described in the instructions) and then sideloaded the magisk variant with adb in recovery.

I started with manual (command line) installation of the same version using factory image.

I'm happy now. Had a couple of problems with root (not all aps are requesting it right), but somehow managed to solve them by making some of them system apps (using magisk module).

persongyu, 2025-04-07:
initial install. im pretty sure ive tried the rootless variant but i will try again when im able to. hope it works!

schnatterer, 2025-04-07:
I am still at a loss why this error started occurring but we have several sources now that confirm the workaround of flashing one variant first and then the other.
Also this mostly seems to happen on initial install, not on OTA updates.

As I do not see anything else I can do right now, I am closing this in the hope you are getting along @persongyu. If not, let me know. I could still create a build without modules.

closed 2025-04-07 by schnatterer

persongyu, 2025-04-08:
it worked!

schnatterer, 2025-04-09:
@mulkamanov "does not work" makes it difficult to help. Maybe if you added some more details on what you tried, the others in this thread can share what they might have done differently to succeed.

mulkamanov, 2025-04-10:
I just wanted to let you know that the workaround may not work for some. I tried different combinations of images, here is the latest:
1. Flashed akita-install-2025040700.zip according to the instructions on the GrapheneOS website.
2. Ran the script to generate patched OTAs. (I commented out the deletion of the temporary directory beforehand).
```
DEVICE_ID=akita MAGISK_PREINIT_DEVICE=sda10 bash -c '. rooted-ota.sh && createRootedOta'
```
There were no error messages in the log, the script completed correctly.
4. Then I followed steps in the "Patch GrapheneOS with OTAs from this image" section of the guide https://github.com/schnatterer/rooted-graphene?tab=readme-ov-file#patch-grapheneos-with-otas-from-this-image
```
❯ ./avbroot ota extract --input rooted-graphene/.tmp/akita-2025040700-970baef-rootless-dirty.zip --directory extracted --fastboot
  0.008s  INFO Extracting from the payload: boot, init_boot, system, vbmeta, vendor_boot
  2.349s  INFO Successfully extracted images from payload

❯ echo $ANDROID_PRODUCT_OUT
extracted

❯ fastboot flashall --skip-reboot
--------------------------------------------
Bootloader Version...: akita-15.2-12862154
Baseband Version.....: g5300o-241205-250127-B-12973597
Serial Number........: 48081JEKB05855
--------------------------------------------

...

Finished. Total time: 76.649s

❯ fastboot reboot-bootloader
Rebooting into bootloader                          OKAY [  0.004s]
Finished. Total time: 0.055s

❯ fastboot erase avb_custom_key
< waiting for any device >
Erasing 'avb_custom_key'                           (bootloader) avb custom key: erase done
OKAY [  0.024s]
Finished. Total time: 0.028s

❯ fastboot flash avb_custom_key grapheneos-ota/avb_pkmd.bin
Warning: skip copying avb_custom_key image avb footer (avb_custom_key partition size: 0, avb_custom_key image size: 1032).
Sending 'avb_custom_key' (1 KB)                    OKAY [  0.001s]
Writing 'avb_custom_key'                           (bootloader) avb custom key: flash done
OKAY [  0.040s]
Finished. Total time: 0.050s
```
5. When trying to reboot in GrapheneOS, the message "Device is corrupt. It can't be trusted" appears. Then phone is stuck on Google logo.

I'll try to figure it out over the weekend, and if I succeed, I'll write here.

schnatterer, 2025-04-10:
Thanks for the details. With these I can give you some hints what to try.
Did you try flashing the rooted image after the rootless one lead to the error?
As you are building your own image, you could try build another one, without modules and the flash it. To do so, build a rootless or rooted image with `SKIP_MODULES=true`. Fingers crossed!

mocarela, 2025-04-11:
IMHO the problem is here: https://issuetracker.google.com/issues/387030988

mulkamanov, 2025-04-11:
@schnatterer thank you for give a hand. I try flashing the rooted image after the rootless one. It also resulted in error. But when I flashed minimal image without any modules, it worked! I was able to boot into the system and install the Magisk.

@mocarela you are probably right. Because I can't lock the bootloader again.
```
❯ fastboot flashing lock
FAILED (remote: 'invalid android images, skip locking')
fastboot: error: Command failed
```
I'll try one of the workarounds listed in the bug report a bit later. Thanks!

schnatterer, 2025-04-12:
@mocarela @mulkamanov Thanks for sharing your insights!
There already were suspicions that there might be a bug in the bootloader https://github.com/schnatterer/rooted-graphene/issues/89#issuecomment-2780039152.

And funny enough flashing the OTA without modules was reported to resolve the issue several times now. But so did flashing rootless image first. Please let me know if you find out anything else or update the README.

ChainSwordCS, 2025-04-18:
I ran into this issue too. I'm on a Google Pixel 6. Similar behavior as mulkamanov experienced [above.](https://github.com/schnatterer/rooted-graphene/issues/96#issuecomment-2794791794) Specifically, upon clicking the power button to go past the "corrupt" screen, then waiting past the other screen (maybe "your bootloader is unlocked", i forget), i get the Google logo and it never advances past that screen. And I checked to see whether fastboot or adb could communicate with the phone when stuck on this screen, they could not. So I had to hold the power button for ~30 seconds each time that happened.

Yes, I also tried just flashing the rootless version instead, and I ran into the same issue.

After reading the two main issue threads, I decided to try the workaround that goes like:
1. install Graphene OS as normal
2. patch GrapheneOS with the OTA (down to and including the step where you flash the avb_custom_key)
3. still within that special Graphene OS fastbootd menu, pick that "Enter recovery" option. and that takes you to a special Graphene OS fastbootd-esque recovery menu...
4. select the option that says "Apply update from ADB"
5. run the command on the PC, for example: `adb sideload ./oriole-2025041100-764135a-rootless.zip`

I tried to find the correct recovery mode menu before, but got tripped up because the normal one just brings me to that screen that says "No command", and I'm not sure how to get to the right menu from there, but I know you can't run `adb sideload` on that screen (even though adb is running).

Anyways, after going through that process it seems at least able to boot into Graphene OS (rootless, but with Custota as a system app or whatever) now. And the boot process shows that "corrupt" screen still, and I have to click the power button to get past it, but _now_ after doing all that stuff it manages to boot normally rather than getting stuck on the Google logo like before.

...

Okay, I wrote all that above first, and now I'm writing this afterwards...

I had a hunch, and decided to give it a try. The first time around, I had done the `fastboot flashall --skip-reboot` and flashed the stuff to slot **a**. And then I think when I did the recovery menu thing and adb sideload, some flag got set (i don't know how the android boot process works :P ) such that when I rebooted to bootloader, it now said "Boot slot: b" instead of "Boot slot: a". 

And I saw in one comment on one of the issues, someone mentioned that the bootloader may throw a tantrum if the avb_custom_key doesn't match what's installed in both slot a and slot b... (For further context: when I got this phone, it was factory reset with Google's stock OS, and I replaced that with regular Graphene and used that for about two months, and now I'm replacing that with this.) 

So I decided to try doing `fastboot flashall --skip-reboot` again, and sure enough, the terminal output indicated it was flashing to slot b this time. Then I repeated the other steps--going to the fastbootd recovery menu, adb sideload (actually using the rooted OTA this time), etc. Except I didn't re-flash the avb_custom_key, as I figured I might not need to, and indeed I did not. 

And now my phone boots normally, without the annoying "corrupt" screen. And I was able to relock the bootloader with no problem (I hadn't tried until just now at the end). Good stuff.

Here are some photos I took of various screens during the process. They might be useful for reference. Feel free to do whatever you want with them, if you find them useful.

| [image] | [image] | [image] | [image] | [image] |
| ------------- | ------------- | ------------- | ------------- | ------------- |
| [image] | [image] | [image] | [image] | [image] |

schnatterer, 2025-04-21:
@ChainSwordCS Thanks for taking the time to share what you did. So basically, you flashed rootless and then sideload the rooted variant. And then you did the same again and it worked without the error?

Some minor hints, I don't know if they help
* Every time you flash or sideload, the new version is applied to the inactive slot. Afterwards, the active slot changes.
* Handling the `No command` screen is described in the Readme

schnatterer, 2025-07-06:
Thanks @farewelltospring for your detailed instructions. I added them to the README.
Did I get it right that you needed to flash and then sideload again to both partitions to succeed?

farewelltospring, 2025-07-20:
Yes, I had to flash the stock image and then sideload the patched OTA twice, into A and B.

I don't know if you can do it in fewer steps but I can confirm that this specific procedure worked for me!

yitzhaq, 2025-07-26:
> Steps I took.
[..]
> Unfortunately root doesn't work but that's a different story ([chenxiaolong/avbroot#462](https://github.com/chenxiaolong/avbroot/issues/462))

So yeah, this hard-bricked my Pixel 8 Pro... :'(

After having tried every other method, resulting in the corrupt message every time (even with rootless), I tried this approach. I had tried the last 3 last builds, together with the corresponding GOS full install zip. Vanilla GOS worked fine each time, but as soon as I tried installing the rooted one, the device refused to boot complaining it was corrupt.

The rooted OTA installed into slot a, so in step 3, I switched to slot b. However, flashing vbmeta failed:

```
% fastboot flashall --skip-reboot
--------------------------------------------
Bootloader Version...: ripcurrent-16.2-13291556
Baseband Version.....: g5300i-250320-250425-B-13407682
Serial Number........: XXXXXXXXXXXXXX
--------------------------------------------
Checking 'product'                                 OKAY [  0.000s]
Setting current slot to 'b'                        OKAY [  0.008s]
Sending 'boot_b' (65536 KB)                        OKAY [  1.580s]
Writing 'boot_b'                                   OKAY [  0.140s]
Sending 'init_boot_b' (8192 KB)                    OKAY [  0.200s]
Writing 'init_boot_b'                              OKAY [  0.024s]
Warning: skip copying vbmeta image avb footer (vbmeta partition size: 0, vbmeta image size: 8192).
Sending 'vbmeta' (8 KB)                            OKAY [  0.001s]
Writing 'vbmeta'                                   FAILED (remote: 'No such file or directory')
```

It wasn't clear from the instructions whether I was supposed to have regenerated the contents of the `extracted` dir with avbroot inbetween flashing each slot, but in any case, I didn't, assuming they would be identical irrespective of which slot they were being installed into.

I then tried rebooting into bootloader first:

```
% fastboot reboot-bootloader     
Rebooting into bootloader                          OKAY [  0.003s]
Finished. Total time: 0.053s
```

Made sure to switch to slot b again:

```
% fastboot --set-active=other    
Setting current slot to 'b'                        OKAY [  0.098s]
```

And then tried anew, this time with seeming success:

```
% fastboot flashall --skip-reboot
--------------------------------------------
Bootloader Version...: ripcurrent-16.2-13291556
Baseband Version.....: g5300i-250320-250425-B-13407682
Serial Number........: XXXXXXXXXXXXXX
--------------------------------------------
Checking 'product'                                 OKAY [  0.000s]
Setting current slot to 'b'                        OKAY [  0.082s]
Sending 'boot_b' (65536 KB)                        OKAY [  1.383s]
Writing 'boot_b'                                   OKAY [  0.084s]
Sending 'init_boot_b' (8192 KB)                    OKAY [  0.174s]
Writing 'init_boot_b'                              OKAY [  0.015s]
Sending 'vbmeta_b' (8 KB)                          OKAY [  0.003s]
Writing 'vbmeta_b'                                 OKAY [  0.006s]
Sending 'vendor_boot_b' (65536 KB)                 OKAY [  1.385s]
Writing 'vendor_boot_b'                            OKAY [  0.085s]
Rebooting into fastboot                            OKAY [  0.000s]
< waiting for any device >
```

However, here things took a turn for the horrible. The device went into a bootloop, never actually entering back into fastboot to finish the rest of the flash. I switched it off to try starting the fastboot manually with vol+power, but now it won't switch on anymore. No combination of buttons switches it back on, even if I hold them for several minutes straight. syslog shows nothing detected if connected while I hold any combination of buttons. Basically everything points to it being completely dead - as if it had zero battery, even though in reality it's fully charged.

This is obviously quite unfortunate, to say the least.. And I am at a loss as to why this happened, and what to do next. Any suggestions would be highly welcome. Can anything be done to bring it back from the dead? Where did I go wrong?

farewelltospring, 2025-07-27:
Sorry to hear what happened to your phone @yitzhaq . I am using a Pixel 9 pro, I don't know if that makes a difference.

FWIW I also remember a similar error when I was doing it and `fastboot flashall --skip-reboot` failed to flash all of the partitions (IIRC it was system in my case, not vbmeta). Here is my shell history immediately after I got confused what was going on. I reset the OS to stock graphene, then flashed the patched image according to avbroot instructions, then sideloaded the patched OTA updates:

```
cd caiman-install-2025061600
sh flash-all.sh
adb shell
adb reboot bootloader
..
ll
fastboot erase avb_custom_key
rm -r extracted
avbroot ota extract --input caiman-ota_update-2025061600.zip.patched --directory extracted --fastboot
fastboot flashall --skip-reboot
fastboot --set-active=other
fastboot flashall --skip-reboot
adb sideload caiman-ota_update-2025061600.zip.patched
adb reboot bootloader
fastboot --set-active=other
fastboot --set-active=a
adb sideload caiman-ota_update-2025061600.zip.patched
adb reboot bootloader
fastboot --set-active=a
adb reboot bootloader
fastboot flash avb_custom_key /path/to/avb_pkmd.bin
adb install Magisk-v29.0.apk
```
I hope you can recover your phone and I hope this helps.

schnatterer, 2025-07-28:
Hi @yitzhaq, 

I am sorry it turned out this way for you. Until now I was not aware there was a scenario that might end in hard-bricking. Unfortunately, I can't offer much help, because I installed all my devices before the device corrupted issue started occurring. I also never had a hard brick case. Are there any guides that help in this case?

Keeping my fingers crossed for you to recover your device.

Thanks @farewelltospring for sharing your thoughts.

yitzhaq, 2025-07-28:
Thanks for both your comments both. Unfortunately, unless by some stroke of miracle, based on my research so far it seems the only way to recover a phone in this state is by some mythical process involving connecting to it over a serial interface using specialized equipment, which very few places (outside of Google engineers or hopefully official Pixel service centers) seem to have. There's talk of some guy in China able to do it also if you can reach him, who offers this as a service - we'll see if i end up chasing this path. Other than that we're probably looking at replacing the board, which is what most run-of-the-mill service centers would likely suggest, and which with the cost of Pixel parts would be prohibitively expensive. Sigh.

As luck would have it (amidst my awful misfortune), I did happen to have another Pixel 8 Pro (long story), which I was planning to sell but now have been forced to switch to using instead. Because of this, and clearly owing to my own foolishness, I continued experimenting with this (pricey experiment so far, this), and have had more luck, so I can report back some of my findings.

First of all, I encountered the exact same problem with corrupt message on this phone by following the official instructions. Not sure if this only happens on newer devices, or only on Pros, or what the pattern is, but what happened was entirely consistent with the previous 8 Pro. I'm starting to suspect this might now be common - perhaps even the norm. FWIW the first device I tried with was running stock 16, the second stock 15.

Instead of flashing both slots (hells no ;), I tried following the instructions, but doing a full sideload of the OTA afterwards. To my joy, this turned out to work just fine. I sure wish I'd tried that first on the first device as well! From my experience so far it does not seem to matter whether you flash the magisk or rootless version first, so if the desired end result is magisk (as I imagine it usually would be), you can probably just go straight to that, and possibly save some steps/downloads. OTA with Custota is also working fine.

So to summarize; the instructions in the README work fine for me *only* if I do an adb sideload of the full OTA before first boot of the system. Otherwise I consistently get the corrupt message, and the device refuses to boot. I don't know (haven't tried) if skipping the avbroot flashing and just sideloading the OTA directly would also work. The exact purpose of going down the avbroot path when you have a full OTA available is still rather unclear to me, but as I'm sure you can imagine, at this point I'm rather hesitant to go too far off-book!

I have also tried [ReZygisk](https://github.com/PerformanC/ReZygisk) and [NeoZygisk](https://github.com/JingMatrix/NeoZygisk), both of which seem to be providing Zygisk support just fine (at least as reported by the Magisk app, as well as by the modules themselves). I imagine this information would be of value to others, so I think this might be worth adding to the README, since Zygisk is not otherwise available out of the box, and it makes rooted life far more livable IMO. I have not (yet?) tried [ZygiskNext](https://github.com/Dr-TSNG/ZygiskNext) as I prefer a fully FOSS implementation, but since the two others are inspired by (and/or are forks) of this one, I imagine that should work just as well.

Now that (painful and costly) experience has shown that hard-bricking is indeed possible, I feel this too merits a big red warning in the README. I imagine where I went wrong in trying to flash to both slots was probably in continuing to try rather than immediately aborting and flashing back stock or vanilla GOS to recover while this might still have been possible. Still, with this at all being a possibility, I would suggest now considering messing with both slots a last resort. Certainly don't want this happening to anyone else.

This is my first hard-brick in probably 10+ years of flashing - and with the most pricey device I've ever (just..) bought, so maybe I've just gotten lucky so far, and got foolhardy because of it. Lesson duly learned. - I hope..!

Also possibly worth noting in the README, though it's perhaps already obvious to most; *DON'T* install any Magisk modules that tamper with the boot image, or really do anything that's not systemless. I accidentally installed one that I thought was systemless, only to find the system immediately unusable on next reboot. Wouldn't boot because integrity was broken, couldn't reflash because bootloader was locked, couldn't unlock it again without wiping. Had to start all over again. Not nearly as bad as the previous SNAFU, and thankfully happened early enough in the setup process that not too much work was lost, but still - not anything you want happening, obviously. Basically I would recommend just being super careful with Magisk modules - not specific to this project, per se, but since presumably the goal of doing this at all is to achieve a rooted system *with* re-locked bootloader, users of this project are all the more susceptible to being hit by this.

All that said, finally I wanna issue a big thank you for this project! I hadn't imagined ever being able to use rooted GOS with locked bootloader (without putting in far more work than I'd consider practical), so your efforts have brought me a big step closer to my ideal setup. Much appreciated - hope you will keep this going for as long as you're able to!

schnatterer, 2025-08-02:
Hi @yitzhaq,

> I'm starting to suspect this might now be common - perhaps even the norm

Yes, from issue reports it seems to me that all devices are getting the corrupted message on first install now. This [seems to have started with version 2025032500](https://github.com/schnatterer/rooted-graphene/issues/89) and the cause is yet to be determined 🤷
Once you succeed it won't occur with updates, though. I flashed all my devices before and have made dozens of OTAs since, all with no problem.

> The exact purpose of going down the avbroot path when you have a full OTA available is still rather unclear to me

That is an interesting find! I acually never challenged those, the instructions came straight from upstream avbroot README.  
So what you are suspecting is:
* flash GOS
* then sideload rooted-graphene
* finally flash `avb_custom_key`

DONE, no `Device is corrupt`?

From your writeup, the most important lessons learned seem to me:
* use sideloading for first installation 
* when flashing fails, **don't switch the slot**

With these, I tried to make the installation instructions more clear: f2458716a03418e71f44c9704feb8f6e2ea80858. 
Do you have anything to add to this?
Maybe some else can verify if sideloading straight will evade the corrupt state altogether.

As for your findings for ReZygisk, NeoZygisk and ZygiskNext, where in the README would you add those? Please feel free to create a PR for this.
I don't use these tools so I won't be the one to add this but I am open for PRs.

Same question for your findings about magisk modules tampering with boot / being not systemless. Where would you expect to find it? There is the warning about OEM unlocking.
Would that be a good place?

Thank you for taking the time to share you extensive LL and for honoring my work on this 😊 I feel bad it turned out so costly for you but am delighted that you can take an overall win from it 👏

yitzhaq, 2025-08-03:
> Yes, from issue reports it seems to me that all devices are getting the corrupted message on first install now. This [seems to have started with version [...]

Yeah, this has been my (admittedly more limited) consistent experience so far, and sounds like it's now apparently the case for everyone.

> That is an interesting find! I acually never challenged those, the instructions came straight from upstream avbroot README. So what you are suspecting is:
>
> * flash GOS
> * then sideload [...]

So this is nothing more than a theory for now, based mostly on speculation (and perhaps ignorance), but that is the working theory, yes. And the ignorance part may honestly be down to a lack of understanding of what avbroot really does and is for. On first glance it *seemed* to me the main purpose of it was simply to allow you to only have to flash the parts that were actually changed, rather than having to apply the full OTA unnecessarily. But then again, isn't that sort of what an OTA *is*, as opposed to a full install? :)  ¯\_(ツ)_/¯

Anyhow, if that is correct, but the more recent developments mentioned above necessitate flashing of the full OTA anyway, then perhaps the installation instructions could just be simplified, while at the same time providing a more reliable working install procedure.

I'd be happy to help test this, but I don't currently have any free Pixel devices to test on. I'll be freeing up a 6 Pro once I complete the migration to the newer 8 Pro, but I'm not there quite yet. I don't see any raven images in your list, though - do you have enough credits to add that device to the list? Would that be much work?

> From your writeup, the most important lessons learned seem to me:
> 
> * use sideloading for first installation
> * when flashing fails, **don't switch the slot**

Yeah, I agree these are the key takeaways. And if the above proves true, I would propose just adjusting (and simplifying) the install instructions accordingly. The flashing process itself would perhaps take a bit longer, but arguably with less prep work, and a more predictable end result.

> With these, I tried to make the installation instructions more clear: [f245871](https://github.com/schnatterer/rooted-graphene/commit/f2458716a03418e71f44c9704feb8f6e2ea80858). Do you have anything to add to this? Maybe some else can verify if sideloading straight will evade the corrupt state altogether.

If someone else wants or is able to help verify, by all means please go ahead :)
The diff looks alright - if I think of anything further to add, I can probably make a PR.

> As for your findings for ReZygisk, NeoZygisk and ZygiskNext, where in the README would you add those? Please feel free to create a PR for this. I don't use these tools so I won't be the one to add this but I am open for PRs.

So in my mind, this would go straight in the notice right at the top saying Zygisk likely will never work, and noting how inconvenient this is :)

That said, however, have you considered bundling @Pixincreate's [Graphene-friendly fork of Magisk](https://github.com/pixincreate/Magisk), rather than the vanilla upstream Magisk (assuming this is what you currently do)? This one tracks Magisk releases, and the only diff against Magisk upstream is the patch they refuse to take which makes Zygisk work on GOS. With this Zygisk works just fine on regular Magisk-rooted GOS, and so I'd be curious to see whether the same would be true for your build. If so, this should simplify the process of getting Zygisk working for people, allow you to remove the warning, and provide an even better experience out of the box.

Happy to help test this too, but as it stands right now, I'd probably need raven builds (plus some time and patience) before I'll be able to help out here.

> Same question for your findings about magisk modules tampering with boot / being not systemless. Where would you expect to find it? There is the warning about OEM unlocking. Would that be a good place?

Yeah, that sounds reasonable. Maybe even a separate headline in that general area.

> Thank you for taking the time to share you extensive LL and for honoring my work on this 😊 I feel bad it turned out so costly for you but am delighted that you can take an overall win from it 👏

You are most welcome - happy if I can help with such a cool project. And even happier if my minor contributions can help make it even better :) As for the unhappy start, I'm mostly glad it seems I can end up with the sort of setup I've wanted - the rest I'm leaving as a future-me-problem for now :o)

schnatterer, 2025-09-27:
Hi @MihkelMK, Thanks for your input!

So what you are saying is that this part of the README was no working advice for you?

> A lot of reports in the issues suggest that the most successful way to avoid it is to follow the instructions below 
until you encounter the error and then sideload the OTA *before* flashing custom keys. 

What worked for you was to 
* follow the README to `fastboot flash avb_custom_key avb_pkmd.bin`,
* then sideloaded the OTA.
* After that you still had `"Device is corrupt. It can't be trusted" `
* You then locked the bootloader  and got rid of the message.

In short: Sideload the OTA *after* flashing the custom keys *and* lock the bootloader.

Correct?
From all the different reports it's rather difficult to describe the one working recipe to success 😵‍💫

MihkelMK, 2025-09-28:
> Hi [@MihkelMK](https://github.com/MihkelMK), Thanks for your input!
>
> So what you are saying is that this part of the README was no working advice for you?
>
>
> What worked for you was to
>
>     * follow [...]

Pretty much. I had two key takeaways:
1. Sideloading OTA _after_ flashing the custom keys and then locking the bootloader lead to a successful install with no corrupt error message
2. Aswer to "Maybe it would be even simpler to just adb sideload the rooted-graphene OTA straight after GrapehenOS." - Did not work for me. `adb sideload` failed to apply rooted-graphene OTA onto stock GrapheneOS. I needed to flash the custom keys for `adb sideload` to work.

ZingyAwesome, 2025-10-01:
I meant to share this a while ago but never got round to it...2 months late isn't too bad, is it?

Sometime around early August I tried to give this a go. I decided to flash on my Pixel Tablet (tangorpro) which was running normal GrapheneOS before doing it on my phone just to make sure it worked first. I forked both the main and OTA repo to add the Pixel Tablet to the device list (uses sda5 as the preinit device) and to use Pixene's fork of Magisk.

After building the releases, I bootloader unlocked my device and got it into fastboot mode without a problem. I was able to follow the README instructions all the way to the `fastboot reboot-bootloader` step. The prior step (fastboot flashall) ran without any errors so I thought everything was going to be fine. However, in my stupidity, I did not remember to to run `fastboot erase avb_custom_key`. Instead, once it had rebooted to the bootloader I tried to boot to system. I remember seeing the "Device is corrupt. It can't be trusted." message while doing this. I then got to the Google logo, which the device got stuck on forever.

At this point I thought, "Okay, no big deal. I can just hold power off, restart and get back to fastboot to erase the avb key and flash my own, then do adb sideload via recovery if that doesn't fix things". However things did not go as planned at all...
After holding power off for ~10 seconds, the screen turned off. And it stayed off. No matter what I did, I could not get the device to power back on. The battery had been at ~80% prior to flashing so that wasn't the issue. I tried every combination of volume and power buttons known to man and nothing happened.

I checked my computer's device manager and noticed the tablet was now displaying as "Pixel ROM recovery". After about an hour of frantic searching, I came to the conclusion that my device had become hard bricked and there was no recovering from the situation. This seemed to be the same situation as @yitzhaq (sorry for mentioning) described:

> However, here things took a turn for the horrible. The device went into a bootloop, never actually entering back into fastboot to finish the rest of the flash. I switched it off to try starting the [...]

After a bit more digging I discovered what this mode actually is. It is somewhat similar to DFU mode on Apple devices, some people call it 'EDL', 'EUB' or 'USBDL'. In this mode, the device _can_ usually be recovered. Only problem is Google think they know what's best for their users and provide 0 way for an end user to do this. In this mode, one can use a serial connection over USB to upload a bootloader to the device to get back into fastboot: only problem is this bootloader file must have 2 special properties. The first is a special bit must be set in the bootloader image that says "this image should be accepted by EUB for recovery". The second is that the image must be signed by Google. In the files Google publicly provides, their bootloader images are signed but do not have the bit set. If you manually set the bit the signature gets invalidated so it is rejected by EUB. There is a whole chain of (rightfully) angry users [complaining about it to Google](https://issuetracker.google.com/issues/402455330) and as you might have expected, Google have done absolutely nothing nor responded to it. They do have these recovery bootloaders but refuse to release them.
Therefore the only way to recover is to hope that these signed recovery bootloaders are leaked. And indeed some of them have been leaked. [This](https://github.com/JoshuaDoes/tensor-usbdl/releases/tag/010) open source tool _might_ help you to recover a hard bricked Pixel 7 series, or 8 series if you haven't installed any update after April. The team behind this tool allegedly has files for more devices which they are privately testing and may become available later. There is a lot more information about this mode and possible recovery from it in a Telegram group run by the same guys behind the tool but I don't want to advertise here, you can find it in their repo.

The Pixel Tablet is based on Pixel 7 so the tool might have saved my device, however I was luckily able to get a replacement under warranty before the tool was released. Would the process have worked if I hadn't been stupid and remembered to erase the avb key? Maybe, but I won't know unless I try again. I would like to try again now that I know there is a recovery tool that may help if I screw up, but it would be an expensive experiment if I end up with another hard brick.

schnatterer, 2025-10-03:
Thanks you all for sharing your experience and thoughts.

@pixincreate Pinning the issue is a great idea. Done. 
I had an exchange with chenxiaolong in April here
https://github.com/schnatterer/rooted-graphene/issues/89#issuecomment-2780039152
Haven't tried GOS devs but can imagine that they might not be too happy to help as rooting via magisk is something they explicitly not support (I don't mean this in disregard).

Did somebody search the GOS forum? 
A quick search yields multiple results:
https://discuss.grapheneos.org/?q=corrupt
I never had the problem so far, so I never researched myself. Just doing my best moderating the many experiences shared and hopefully helping others avoid bricking 😬
Nice little community effort ❤️

@MihkelMK Thanks for getting back to me regarding the question on the Readme. Time to update it. Do you think @ZingyAwesome's 10 steps could work? I am considering putting those in the Readme. 

@ZingyAwesome Thanks for documenting the steps you took and mentioning 
https://github.com/JoshuaDoes/tensor-usbdl/!

ZingyAwesome, 2025-10-03:
Just tried the same 10 steps on a Pixel 9 (tokay) and it worked too 🎉. Again, using the method I did not see a "Device is corrupt." message.
I think it would be great to update the README with some instructions that specifically mention the correct order in which the locking/unlocking, flashing, avb key operations and sideloading should be done.