schnatterer/rooted-graphene #104: stuck on G logo

Moortu, 2025-04-27:
Hi,

I followed the instructions to the letter.
I got version caiman-install-2025041100.zip from grapheneOS since the current release of rooted-graphene isn't up to date yet.
I manually flashed it and booted into it, everything fine.

Then I got caiman-2025041100-764135a-magisk-v28.1.zip
and followed the instructions again

I try the optional step to boot into it before locking the bootloader, and there it just gets stuck on the G logo

my install logs below.

` platform-tools  tar xvf .\caiman-install-2025041100.zip
x caiman-install-2025041100/radio-caiman-g5400c-241205-250127-b-12971716.img
x caiman-install-2025041100/bootloader-caiman-ripcurrentpro-15.2-13237002.img
x caiman-install-2025041100/d3m2.ec.bin
x caiman-install-2025041100/evt.ec.bin
x caiman-install-2025041100/proto11.ec.bin
x caiman-install-2025041100/avb_pkmd.bin
x caiman-install-2025041100/android-info.txt
x caiman-install-2025041100/android-info.zip
x caiman-install-2025041100/boot.img
x caiman-install-2025041100/init_boot.img
x caiman-install-2025041100/dtbo.img
x caiman-install-2025041100/vendor_kernel_boot.img
x caiman-install-2025041100/pvmfw.img
x caiman-install-2025041100/vendor_boot.img
x caiman-install-2025041100/vbmeta.img
x caiman-install-2025041100/super_1.img
x caiman-install-2025041100/super_2.img
x caiman-install-2025041100/super_3.img
x caiman-install-2025041100/super_4.img
x caiman-install-2025041100/super_5.img
x caiman-install-2025041100/super_6.img
x caiman-install-2025041100/super_7.img
x caiman-install-2025041100/super_8.img
x caiman-install-2025041100/super_9.img
x caiman-install-2025041100/super_10.img
x caiman-install-2025041100/super_11.img
x caiman-install-2025041100/super_12.img
x caiman-install-2025041100/super_13.img
x caiman-install-2025041100/super_14.img
x caiman-install-2025041100/script.txt
x caiman-install-2025041100/flash-all.sh
x caiman-install-2025041100/flash-all.bat
 platform-tools  cd .\caiman-install-2025041100\
 caiman-install-2025041100  .\flash-all.bat
Available devices:
49241FDAP000NC         fastboot
Warning: skip copying bootloader_b image avb footer (bootloader_b partition size: 0, bootloader_b image size: 18540472).
Sending 'bootloader_b' (18105 KB)                  OKAY [  0.046s]
Writing 'bootloader_b'                             (bootloader) Flashing pack version ripcurrentpro-15.2-13237002
(bootloader) flashing platform ripcurrentpro
(bootloader) Validating partition ufs
(bootloader) Validating partition ufs
(bootloader) Validating partition partition:0
(bootloader) Validating partition partition:1
(bootloader) Validating partition partition:2
(bootloader) Validating partition partition:3
(bootloader) Validating partition bl1_b
(bootloader) Validating partition pbl_b
(bootloader) Validating partition bl2_b
(bootloader) Validating partition abl_b
(bootloader) Validating partition bl31_b
(bootloader) Validating partition tzsw_b
(bootloader) Validating partition gsa_b
(bootloader) Validating partition gsa_bl1_b
(bootloader) Validating partition ldfw_b
(bootloader) Validating partition gcf_b
[... 284 lines cut ...]
Writing 'system_a'                                 OKAY [  0.410s]
Sending sparse 'system_a' 4/6 (262116 KB)          OKAY [  0.946s]
Writing 'system_a'                                 OKAY [  0.399s]
Sending sparse 'system_a' 5/6 (255096 KB)          OKAY [  0.903s]
Writing 'system_a'                                 OKAY [  0.367s]
Sending sparse 'system_a' 6/6 (20736 KB)           OKAY [  0.098s]
Writing 'system_a'                                 OKAY [  0.047s]
Finished. Total time: 47.811s
 platform-tools  fastboot reboot-bootloader
Rebooting into bootloader                          OKAY [  0.002s]
Finished. Total time: 0.004s
 platform-tools  fastboot devices
49241FDAP000NC   fastboot
 platform-tools  fastboot erase avb_custom_key
Erasing 'avb_custom_key'                           (bootloader) avb custom key: erase done
OKAY [  0.025s]
Finished. Total time: 0.029s
 platform-tools  curl -s https://raw.githubusercontent.com/rooted-graphene/ota/refs/heads/main/avb_pkmd.bin > avb_pkmd.bin
 platform-tools  fastboot flash avb_custom_key avb_pkmd.bin
Warning: skip copying avb_custom_key image avb footer (avb_custom_key partition size: 0, avb_custom_key image size: 1032).
Sending 'avb_custom_key' (1 KB)                    OKAY [  0.001s]
Writing 'avb_custom_key'                           (bootloader) avb custom key: flash done
OKAY [  0.042s]
Finished. Total time: 0.051s
 Kris   platform-tools                                        `

Moortu, 2025-04-27:
see others for solutions

Moortu, 2025-04-28:
I tried again with the just released caiman-2025042500-5938466-magisk-v28.1.zip
Nothing, still just says it's corrupt and if I try to boot gets stuck again on G logo.

Also I used linux this time to take that out of the equation

Moortu, 2025-04-28:
And finally tried with caiman-2025042500-5938466-rootless.zip.
Still the same.

Only thing I see everytime is : 
Warning: skip copying avb_custom_key image avb footer (avb_custom_key partition size: 0, avb_custom_key image size: 1032).

schnatterer, 2025-04-28:
The warning seems to be normal: https://github.com/schnatterer/rooted-graphene/issues/89#issuecomment-2777184697

I released `minimal` variants (without custota) for your device. These normally worked when everything else failed. If they work you should be able to sideload the normal magisk variants afterwards.

Moortu, 2025-04-28:
> The warning seems to be normal: [#89 (comment)](https://github.com/schnatterer/rooted-graphene/issues/89#issuecomment-2777184697)
> 
> I released `minimal` variants (without custota) for your device. These normally worked when everything else failed. If they work you should be able to sideload the normal magisk variants afterwards.

so I used the instruction https://github.com/schnatterer/rooted-graphene/issues/96#issuecomment-2814819354
and it worked.
 I can now boot into grapheneos and I have custota.

Is it possible that the flash all --skip-reboot command might need the --slot=all option?

Moortu, 2025-04-28:
to build and refine the steps to get it to work:

1. install Graphene OS as normal
2. patch GrapheneOS with the OTA (down to and including the step where you flash the avb_custom_key, you will be in fastboot now)
3. fastboot reboot fastboot  (you will be returned to fastbootd)
4. recovery mode
5. Apply update from ADB
6. in your terminal: adb sideload ./rootless.zip

Moortu, 2025-04-28:
... I just bricked my device.

Moortu, 2025-04-28:
to build upon and refine the steps, from ChainSwordCS, to get it to work:

WARNING: I CAME CLOSE TO BRICKING MY DEVICE

    install Graphene OS as normal
    patch GrapheneOS with the OTA (down to and including the step where you flash the avb_custom_key, you will be in fastboot now)
    in your terminal: fastboot reboot fastboot (you will be returned to fastbootd, you'll see the G first, no worries)
    recovery mode
    Apply update from ADB
    in your terminal: adb sideload ./rootless.zip
    reboot into system
    in custato set the ota link to https://rooted-graphene.github.io/ota/magisk (or whatever you require)
    install update
    enable usb debugging and lock the bootloader according to the guide

Moortu, 2025-04-28:
@schnatterer the minimal zip worked!

schnatterer, 2025-04-28:
That's great! I still wonder why... it would be awkward to also release a minimal zip only for Bootstrapping 😐 b

Did the instructions from issue 96 not work after all?

Moortu, 2025-04-29:
@schnatterer 
I think the mistake I made was, in magisk using the install magisk again. causing a system that cannot be booted.

So the instructions probably do work.
But it's a difficult route to walk

closed 2025-05-03 by Moortu