schnatterer/rooted-graphene #119: Add one of those "Zygisk-capable-on-graphene" magisk mutations as one of alternate custota links please (need xposed for a cool thing)

Aleximal, 2025-05-25:
Hello!
Thanks for the awesome project!
I did read the readme and did notice Zygisk requires an altered magisk fork on Graphene

I do understand that enabling it as part of "rooted Graphene offering" would consume resources and Xposed (for which I need Zygisk) is even a bigger problem, but turns out there is one (1) Xposed module that offers a truly interesting capability 

https://github.com/icepony/AlwaysCreateUser

As it stands Graphene, for no human relatable reason, still constrains the number of work profiles to 1 (so the only option is to get 1 work profile and 1 private space, which I emphatically do not enjoy for several reasons) 

And it would appear this humble Xposed module should be able to solve this injustice (theoretically)

schnatterer, 2025-05-26:
I understand why some users might want to have this. But I won't be the one to maintain it. My time is limited, and I already do much more than supporting the few devices I own.

Did you have a look https://github.com/pixincreate/PixeneOS?

What I can offer you, though would be an option to pass in an apk URL of your choice, so you could easily maintain your own version of rooted-graphene using a different magisk fork.
With this, you could just fork https://github.com/rooted-graphene/ota and configure the URL to your magik fork. 

 I would even list your fork in the Readme, once you set it up.

Aleximal, 2025-05-27:
Thanks! That's fair!