schnatterer/rooted-graphene #130: How to add a version of fdroid privileged extension for the play store in the compile?

ixanza, 2025-06-17:
I have my own fork of the building repo however I'm not sure how to add a version of fdroid privileged extension rebuilt with play store in it to give it unattended installations into the compile so I don't need root similar to how you did custota.

I'm following the guide on their gitlab https://gitlab.com/fdroid/privileged-extension
to rebuild the apk with com.google.vending in the whitelist 

Reason for this is my work profile doesn't allow the installation dialog so I have to somehow make play store do unattended installations when triggered by my work app. 

Alternative would be to push play store to priv app but that would defeat the purpose of sandboxed google play.

schnatterer, 2025-06-20:
I actually thought about adding fdroid priv extension and the aurora services as well because they provide a simple way to automate app updates without root access.
Both tools seem to be deprecated or unmaintained, though.

Anyways, the way to go would be a patch.py python module like the one for [Custota](https://github.com/chenxiaolong/my-avbroot-setup/blob/master/lib/modules/custota.py).

schnatterer, 2025-08-03:
Is you'll ever make any progress on this, I'd be interested in how it turned out.

closed 2025-08-03 by schnatterer